arduino/ArduinoTensorFlowLiteTutorials #7: Model Is Not Building

darcstartech, 2019-11-22:
@lucach
@8bitkick 
@sandeepmistry 

Hi!  

I have created the attached .csv files (currently .txt so I can upload to GitHub) for the punch and flex gestures and cannot get them to create a model.  It seems to just generate nothing.  Not sure what I am doing wrong? 

[punch.txt](https://github.com/arduino/ArduinoTensorFlowLiteTutorials/files/3878271/punch.txt)
[flex.txt](https://github.com/arduino/ArduinoTensorFlowLiteTutorials/files/3878273/flex.txt)

[image]

lucach, 2019-11-22:
Hi! I think it is probably due to the fact that you have "null"/"empty" values in your record set (double check what you are copy-pasting from the serial monitor, it shouldn't be the case).

```sh
> grep ,, flex.txt
,,,,,,
,,,,,,
,,,,,,
,,,,,,
,,,,,,
,,,,,,
,,,,,,
,,,,,,
,,,,,,
,,,,,,
```

darcstartech, 2019-11-30:
Awesome, thank you!  That solved the issue.  

This is a great project, thank you for doing this and releasing it online. So cool!

closed 2019-11-30 by darcstartech