arduino/ArduinoTensorFlowLiteTutorials #10: Not trainning the model

JoaquinRomeroPrada, 2020-03-23:
Hi!

I'm a beginner with TensorFlow and I'm currently trying to adapt your model to one that can sort falls.

My dataset consist in a 119 falls with 300 samples each, taken with an accelerometer with a range of +-16g and +-2000º (so I've changed te parameters given to normalize the dataset). 

I get something like this every time I run the "Build & Train the Model" part on your code:
[image]

And I'm not sure what to do to make it work correctly...

PrashantDandriyal, 2020-04-06:
Hi @JoaquinRomeroPrada What was the sample rate at which your accelerometer (sensor) reads data ? 
It really matters. Or your data may not make any sense to the model. Suppose your sensor takes in data @5Hz i.e., 5 sets of acc_x, acc_y and acc_z per second (a total of 15 values). In this case, you should be feeding your model a 2D data, having a row dimension = 5 x 3 or 3 x 5. And if you have 119 falls, the overall data should look something like 119 x 3 x 5 or 119 x 5 x 3. BTW, I didn't get what you meant by 300 samples each.... 
I implemented something similar using the MotionSense Dataset, I think it may help https://github.com/PrashantDandriyal/arduino_GSoC2020/blob/master/activity_tracker/activity_tracker_motionSense_gsoc20.ipynb

JoaquinRomeroPrada, 2020-04-06:
Hi @PrashantDandriyal

My accelerometer is reading at 100Hz, but the idea is to get a "window" with 300 samples in this case or more (I would have to test it) and feed the 300 samples (1800 values from 3D accel and 3D gyro) to the model every X miliseconds so it could detect if a fall has occured.

However, to train the model I've used the SisFall dataset which is obtained at 200Hz. The idea was to take 119 falls, get the most representative 300 samples of each one of them (so if a fall has, let's say 10000 samples, I would take the 300 when the fall is happening), train the model with them and do the described on the first paragraph.

I had your code to work by not just adding Falls to the trainning set, but a combination of Falls and ADLs. So in the end, the trainning data would look like a couple of .csv files with 50 rows each, one for Falls and one for ADLs, with 1800 values in each cell (300 aX, 300 aY...).

PrashantDandriyal, 2020-04-07:
So, is the issue resolved ?

closed 2020-04-07 by JoaquinRomeroPrada