arduino/ArduinoTensorFlowLiteTutorials #11: Tiny ML on Arduino: Gesture recognition tutorial results in a blank model.h

jomoengineer, 2020-04-05:
I attempted to run through the Tiny ML on Arduino: Gesture recognition example and it resulted in a a model.h file that is just a model array declaration. 

Ex:
```
const unsigned char model[] = {
};

```

This is the example that I was following:
[https://colab.research.google.com/github/arduino/ArduinoTensorFlowLiteTutorials/blob/master/GestureToEmoji/arduino_tinyml_workshop.ipynb#scrollTo=f92-4Hjy7kA8](url)

The following Warnings were seen while running the example:
NOTE: An issue was filed with the Colab project but I do not see it listed.

Also, a gesture_model.tflite file was created and has about 145K of data.

- Setup Python Environment

```
E: Failed to fetch http://security.ubuntu.com/ubuntu/pool/main/v/vim/xxd_8.0.1453-1ubuntu1.1_amd64.deb  404  Not Found [IP: 91.189.88.152 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

```

  - Build & Train the Model
```
Train on 12 samples, validate on 4 samples
Epoch 1/600
WARNING:tensorflow:Entity <function Function._initialize_uninitialized_variables.<locals>.initialize_variables at 0x7fe172912ae8> could not be transformed and will be executed as-is. Please report this to the AutoGraph team. When filing the bug, set the verbosity to 10 (on Linux, `export AUTOGRAPH_VERBOSITY=10`) and attach the full output. Cause: module 'gast' has no attribute 'Num'
WARNING: Entity <function Function._initialize_uninitialized_variables.<locals>.initialize_variables at 0x7fe172912ae8> could not be transformed and will be executed as-is. Please report this to the AutoGraph team. When filing the bug, set the verbosity to 10 (on Linux, `export AUTOGRAPH_VERBOSITY=10`) and attach the full output. Cause: module 'gast' has no attribute 'Num'
12/12 [==============================] - 1s 50ms/sample - loss: 0.3119 - mae: 0.5501 - val_loss: 0.2502 - val_mae: 0.4998
Epoch 2/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2653 - mae: 0.5116 - val_loss: 0.2498 - val_mae: 0.4994
Epoch 3/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2620 - mae: 0.5116 - val_loss: 0.2493 - val_mae: 0.4991
Epoch 4/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2628 - mae: 0.5082 - val_loss: 0.2488 - val_mae: 0.4985
Epoch 5/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2569 - mae: 0.5058 - val_loss: 0.2443 - val_mae: 0.4942
Epoch 6/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2553 - mae: 0.4992 - val_loss: 0.2579 - val_mae: 0.4993
Epoch 7/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2619 - mae: 0.5072 - val_loss: 0.2531 - val_mae: 0.4989
Epoch 8/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2576 - mae: 0.5051 - val_loss: 0.2521 - val_mae: 0.4983
Epoch 9/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2569 - mae: 0.5049 - val_loss: 0.2499 - val_mae: 0.4981
Epoch 10/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2544 - mae: 0.5036 - val_loss: 0.2491 - val_mae: 0.4979
Epoch 11/600
12/12 [==============================] - 0s 3ms/sample - loss: 0.2533 - mae: 0.5029 - val_loss: 0.2485 - val_mae: 0.4978
```

- Run with Test Data
```
/usr/local/lib/python3.6/dist-packages/ipykernel_launcher.py:10: MatplotlibDeprecationWarning: cycling among columns of inputs with non-matching shapes is deprecated.
  # Remove the CWD from sys.path while we load stuff.
/usr/local/lib/python3.6/dist-packages/ipykernel_launcher.py:11: MatplotlibDeprecationWarning: cycling among columns of inputs with non-matching shapes is deprecated.
  # This is added back by InteractiveShellApp.init_path()

```

- Convert the Trained Model to Tensor Flow Lite
```
WARNING:tensorflow:Entity <function Function._initialize_uninitialized_variables.<locals>.initialize_variables at 0x7fe16a17d598> could not be transformed and will be executed as-is. Please report this to the AutoGraph team. When filing the bug, set the verbosity to 10 (on Linux, `export AUTOGRAPH_VERBOSITY=10`) and attach the full output. Cause: module 'gast' has no attribute 'Num'
WARNING: Entity <function Function._initialize_uninitialized_variables.<locals>.initialize_variables at 0x7fe16a17d598> could not be transformed and will be executed as-is. Please report this to the AutoGraph team. When filing the bug, set the verbosity to 10 (on Linux, `export AUTOGRAPH_VERBOSITY=10`) and attach the full output. Cause: module 'gast' has no attribute 'Num'
Model is 147764 bytes
```

jomoengineer, 2020-04-05:
I found a temp solution.  
I copied the "Encode the Model in an Arduino Header File" code and the gesture_model.tflite file to a Linux box, created a content folder and made a shell and a python script to run the code.
Ex:
- create_content.sh
```
#!/bin/bash

echo "const unsigned char model[] = {" > content/model.h
cat gesture_model.tflite | xxd -i      >> content/model.h
echo "};"                              >> content/model.h
```

- get_header_size.py

```
#!/usr/bin/env python3

import os
model_h_size = os.path.getsize("./content/model.h")
print(f"Header file, model.h, is {model_h_size:,} bytes.")
print("\nOpen the side panel (refresh if needed). Double click model.h to download the file.")

```

I then copied the resulting model.h code to the Arduino IDE project and was able to complete the Gesture recognition tutorial. 

This worked for the "TinyML Classify objects by color" tutorial as well.

PrashantDandriyal, 2020-04-06:
Please refer to this example on Google Collab. I hope it helps https://github.com/PrashantDandriyal/arduino_GSoC2020/blob/master/activity_tracker/activity_tracker_motionSense_gsoc20.ipynb

jomoengineer, 2020-04-06:
Uh, the link you provided is a GitHub link.  I'm not see how that address this issue.

PrashantDandriyal, 2020-04-07:
It is actually a link to the iPython notebook I shared to demonstrate how I obtained the header file-model and thought it may help you get over your errors.

intratron, 2020-09-13:
I got the same problem. The model.h file was blank. 

I was able to solve it. What I did was to verify the file using TextEdit on a Mac.  I realized some of the lines/cells of the flex.csv and punch.csv files where either empty, mixed with another cell or a whole line of cells was out. See the line:

This is ok:
-0.37 | 0.395 | 1.931 | 284.363 | -9.46 | -77.026
-0.648 | 0.578 | 1.737 | 795.166 | 36.743 | -190.247
-0.238 | 0.815 | 1.104 | 1028.137 | 40.283 | -246.948

This is not ok:
-0.37 | 0.395 | 1.931 | **284.363-9.46** | **(empty)** | -77.026
-0.648 | 0.578 | 1.737 | 795.166 | 36.743 | -190.247 **-0.238 | 0.815 | 1.104 | 1028.137 | 40.283 | -246.948**

So I had to open the file in excel and manually fix every error in the cells, I took me like 5 minutes but that was the only way I managed to run the colab webpage and run the training, convert and encode functions. After doing this, the model.h file was generated with the data (911kb).

Not everything was ok though. I ran the sketch in Arduino IDE using the  IMU_Classifier, adding a tab, pasting the model.h data in it. The Arduino serial monitor is responding but is giving me just a "nan" label instead of a float eg. 0.8987

So apparently I managed to solve the model.h problem but still I don't know whats the "nan" problem.

JatinPendharkar, 2021-06-28:
@intratron im getting the same error. Did you find any solution for same ?
The Arduino serial monitor is responding but is giving me just a "nan" label instead of a float eg. 0.8987