arduino/ArduinoTensorFlowLiteTutorials #12: No official Repository of Library

PrashantDandriyal, 2020-04-06:
Hi, 
I am applying to Arduino this season of GSoC 2020. So, I would like to contribute to the **Arduino_TensorFlowLite** libarry, but I fail to find the upstream on GitHub. Can you please tell me the reason behind that ?

per1234, 2020-04-06:
I believe the upstream is here:
https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro

Arduino's mirror of TensorFlow Lite formatted as an Arduino library is here:
https://github.com/bcmi-labs/tensorflow_lite_mirror

PrashantDandriyal, 2020-04-07:
Thanks @per1234 
But why isn't it a part of the official libraries? I proposed to add examples to this library (referring to the TensorFlow upstream) but I wonder if it the library is ineligible for that right now...

per1234, 2020-04-07:
>but the mirror site seems broken !

Sorry, I didn't notice it's a private repository. That's why you get a 404 error from that URL. The repository being private is not too much of a problem since it's only a mirror.

>why isn't it a part of the official libraries?

It is official. bcmi-labs is one of the GitHub organizations owned by Arduino. I'd guess it's not a public repository under the arduino-libraries organization because that would lead to people submitting issue reports and pull requests that should have been directed to the upstream repository instead.

>I wonder if it the library is ineligible for that right now

I don't know the answer to that. You could ask in your dedicated issue in the [arduino/summer-of-code repository](https://github.com/arduino/summer-of-code).

PrashantDandriyal, 2020-04-07:
So, the only means of contribution is the TensorFlow repo. Right ?

per1234, 2020-04-07:
Yes. It wouldn't make sense to contribute to a mirror, since the whole point of a mirror is to follow the upstream. So you make changes to the upstream repository and then those changes eventually get pulled into the mirror repository.

Of course, you have the option of contributing to the tutorial content in this repository.

PrashantDandriyal, 2020-04-07:
Thank you very much :)

closed 2020-04-07 by PrashantDandriyal