arduino/ArduinoTensorFlowLiteTutorials #13: How can I add BLE to this code?

schosdas, 2020-04-10:
**Hello. Thanks to you, I learned how to use tensorflow. And I added a count of gestures to this code. Now I'm going to send this number of counts to my smartphone via BLE. 
However, when you add  BLE code, it will be uploaded but the Bluetooth connection will not work.
This is the full code I wrote. I would really appreciate it if you could tell me the wrong thing.**

```
#include <ArduinoBLE.h>
#include <Arduino_LSM9DS1.h> 
  
#include <TensorFlowLite.h>
#include <tensorflow/lite/experimental/micro/kernels/all_ops_resolver.h>
#include <tensorflow/lite/experimental/micro/micro_error_reporter.h>
#include <tensorflow/lite/experimental/micro/micro_interpreter.h>
#include <tensorflow/lite/schema/schema_generated.h>
#include <tensorflow/lite/version.h>

#include "model.h"

BLEService yscheulService("00001101-0000-1000-8000-00805F9B34FB"); 
BLEStringCharacteristic LevelChar("00002a11-0000-1000-8000-00805f9b34fb", BLERead | BLENotify,40);

const float accelerationThreshold = 2.5; // threshold of significant in G's
const int numSamples = 119;

int samplesRead = numSamples;

// global variables used for TensorFlow Lite (Micro)
tflite::MicroErrorReporter tflErrorReporter;

// pull in all the TFLM ops, you can remove this line and
// only pull in the TFLM ops you need, if would like to reduce
// the compiled size of the sketch.
tflite::ops::micro::AllOpsResolver tflOpsResolver;

const tflite::Model* tflModel = nullptr;
tflite::MicroInterpreter* tflInterpreter = nullptr;
TfLiteTensor* tflInputTensor = nullptr;
TfLiteTensor* tflOutputTensor = nullptr;

// Create a static memory buffer for TFLM, the size may need to
// be adjusted based on the model you are using
constexpr int tensorArenaSize = 8 * 1024;
byte tensorArena[tensorArenaSize];

// array to map gesture index to a name
const char* GESTURES[] = {"flex"};

#define NUM_GESTURES (sizeof(GESTURES) / sizeof(GESTURES[0]))

int count=0; // gesture count

//********************************************************************************************************************************

void setup() {
 Serial.begin(9600);
  if (!IMU.begin()) { //LSM9DS1센서 시작
    Serial.println("LSM9DS1 failed!!");
    while (1);
  }

  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");
    while (1);
  }

  // print out the samples rates of the IMUs
  Serial.print("Accelerometer sample rate = ");
  Serial.print(IMU.accelerationSampleRate());
  Serial.println(" Hz");
  Serial.print("Gyroscope sample rate = ");
  Serial.print(IMU.gyroscopeSampleRate());
  Serial.println(" Hz");
  Serial.println();

  // get the TFL representation of the model byte array
  tflModel = tflite::GetModel(model);
  if (tflModel->version() != TFLITE_SCHEMA_VERSION) {
    Serial.println("Model schema mismatch!");
    while (1);
  }
  // Create an interpreter to run the model
  tflInterpreter = new tflite::MicroInterpreter(tflModel, tflOpsResolver, tensorArena, tensorArenaSize, &tflErrorReporter);
  
  // Allocate memory for the model's input and output tensors
  tflInterpreter->AllocateTensors();
  
  // Get pointers for the model's input and output tensors
  tflInputTensor = tflInterpreter->input(0);
  tflOutputTensor = tflInterpreter->output(0);

   // set advertised local name and service UUID:
  BLE.setAdvertisedService(yscheulService); // add the service UUID
  yscheulService.addCharacteristic(LevelChar);
  BLE.addService(yscheulService);
  // start advertising
  BLE.setLocalName("yschul");
  // start advertising
  BLE.advertise(); 
}
//********************************************************************************************************************************
String Level_String;
float aX, aY, aZ, gX, gY, gZ;

void updategyroLevel() {
//  if (IMU.accelerationAvailable()) {
//    IMU.readAcceleration(aX, aY, aX);
//    IMU.readGyroscope(gX, gY, gZ);
//    }
  Level_String=String(a); // count value -> BLE
  LevelChar.writeValue(Level_String);
  Serial.println(Level_String);
}
//********************************************************************************************************************************
void loop() {
  // wait for significant motion
  while (samplesRead == numSamples) {
    if (IMU.accelerationAvailable()) {
      // read the acceleration data
      IMU.readAcceleration(aX, aY, aZ);

      // sum up the absolutes
      float aSum = fabs(aX) + fabs(aY) + fabs(aZ);

      // check if it's above the threshold
      if (aSum >= accelerationThreshold) {
        // reset the sample read count
        samplesRead = 0;
        break;
      }
    }
  }
  // check if the all the required samples have been read since
  // the last time the significant motion was detected
  while (samplesRead < numSamples) {
    // check if new acceleration AND gyroscope data is available
    if (IMU.accelerationAvailable() && IMU.gyroscopeAvailable()) {
      // read the acceleration and gyroscope data
      IMU.readAcceleration(aX, aY, aZ);
      IMU.readGyroscope(gX, gY, gZ);

      // normalize the IMU data between 0 to 1 and store in the model's
      // input tensor
      tflInputTensor->data.f[samplesRead * 6 + 0] = (aX + 4.0) / 8.0;
      tflInputTensor->data.f[samplesRead * 6 + 1] = (aY + 4.0) / 8.0;
      tflInputTensor->data.f[samplesRead * 6 + 2] = (aZ + 4.0) / 8.0;
      tflInputTensor->data.f[samplesRead * 6 + 3] = (gX + 2000.0) / 4000.0;
      tflInputTensor->data.f[samplesRead * 6 + 4] = (gY + 2000.0) / 4000.0;
      tflInputTensor->data.f[samplesRead * 6 + 5] = (gZ + 2000.0) / 4000.0;

      samplesRead++;

      if (samplesRead == numSamples) {
        // Run inferencing
        TfLiteStatus invokeStatus = tflInterpreter->Invoke();
        if (invokeStatus != kTfLiteOk) {
          Serial.println("Invoke failed!");
          while (1);
          return;
        }

        // Loop through the output tensor values from the model
        Serial.print(GESTURES[0]); // GESTURES[0] = flex
        Serial.print(": ");
        Serial.println(tflOutputTensor->data.f[0], 6);       
        Serial.println();  
      }
    }
  }
// Counts when accuracy is 0.7 or higher
 if(tflOutputTensor->data.f[0] >= 0.7)
        {
          count=count+1;
          Serial.println(count);
        }
  
  // listen for BLE peripherals to connect:
  BLEDevice central = BLE.central();
  // if a central is connected to peripheral:
  if (central) {
    while (central.connected()) {
        updateyscheulLevel();
      }
    } 
}

```

per1234, 2020-04-10:
Related forum discussion:
https://forum.arduino.cc/index.php?topic=676353

8bitkick, 2020-09-23:
This is a user BLE implementation issue covered in the forum link above. I think we can close the issue here as it is both resolved in the forum and unrelated to the tutorial itself.  cc @per1234

closed 2020-09-23 by 8bitkick