arduino/ArduinoTensorFlowLiteTutorials #15: tensorflow/lite/experimental/micro/kernels/all_ops_resolver.h: No such file or directory

wallanoo, 2020-06-23:
When I upload the IMU_Classifier to the Arduino nano 33 sense BLE I obtain this error:

tensorflow/lite/experimental/micro/kernels/all_ops_resolver.h: No such file or directory

I have the Arduino_Tensorflowlite library 2.1.0-Alpha precompiled installed..

wallanoo, 2020-06-23:
I found libraries in tensorflow/lite/micro

closed 2020-06-23 by wallanoo

myominhtet, 2020-07-22:
how get that. I tried but still got error

jintolonappan, 2020-08-07:
@myominhtet 
You may need to change such lines to:

```
#include "tensorflow/lite/micro/all_ops_resolver.h"
#include "tensorflow/lite/micro/micro_error_reporter.h"
#include "tensorflow/lite/micro/micro_interpreter.h"
#include "tensorflow/lite/schema/schema_generated.h"
#include "tensorflow/lite/version.h"
```

pdp7, 2020-09-07:
I ran into this as well.

I had to remove 'experimental' from the paths:
```
#include <tensorflow/lite/micro/all_ops_resolver.h>
#include <tensorflow/lite/micro/micro_error_reporter.h>
#include <tensorflow/lite/micro/micro_interpreter.h>
```