arduino/ArduinoTensorFlowLiteTutorials #23: COM stuck after uploading IMU Classifier sketch？

KXCY-AI, 2020-10-20:
New to Arduino, I follow with the Arduino TensorFlow Lite Tutorials and trained "GestureToEmoji" IMU Classifier model, which is deployed on BLE. It  worked well previously, yet its stuck a serial monitor these day, after upload sketch to the board. Where I can figure out more information about it?

per1234, 2020-10-20:
I'm working on getting enough information to be able to investigate this issue in @KXCY-AI's Arduino Forum post about the same:
https://forum.arduino.cc/index.php?topic=709946

KXCY-AI, 2020-10-20:
> I'm working on getting enough information to be able to investigate this issue in @KXCY-AI's Arduino Forum post about the same:
> https://forum.arduino.cc/index.php?topic=709946
> 
> Also posted at [arduino/ArduinoCore-nRF528x-mbedos#106](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/106)

Thanks per1234, the model runs well after downgrade to 1.1.6

gtteixeira, 2020-11-25:
I am getting the same problem. Follow all the steps with success on training and test model using the colab. No problem. Download the model.h exactly as instructions. Compiled ok.

 But when you ask the serial monitor nothing happen. Completely freezing. I am trying to fix during 4 days with no success.

model file and compilation was ok:
[image]

Freezing when call the serial monitor:
[image]

I am using tensorflow 2.1.0
[image]

Rainycat, 2020-12-18:
We've hit the exact same problem.  Adding some debugging with
Serial.println(), we can see that it never returns from the call to
"new tflite::MicroInterpreter()".

Arduino IDE version is 1.8.13

Arduino BLE 33 board support version is 1.1.6

Tensorflow library version is 2.1.0. ALPHA (IMU_Classifier failed to build against older versions)

Hope this helps someone narrow down the problem!

BenLuchterhand, 2021-01-01:
> I'm working on getting enough information to be able to investigate this issue in @KXCY-AI's Arduino Forum post about the same:
> https://forum.arduino.cc/index.php?topic=709946
> 
> Also posted at [arduino/ArduinoCore-nRF528x-mbedos#106](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/106)

Hey, I'm new to Arduino so I apologize if this is a silly question, but how do I downgrade the core? I'm experience the same issue.

Rainycat, 2021-01-02:
> We've hit the exact same problem. Adding some debugging with
> Serial.println(), we can see that it never returns from the call to
> "new tflite::MicroInterpreter()".

Just to update anyone else who is having this problem, after a lot of faff, we discovered that in MicroAllocator (called by MicroInterpreter instantiation), it checks the supplied arena is 16-byte aligned and errors out if it isn’t.  The fix is to declare the arena like this:

byte tensorArena[tensorArenaSize] __attribute__((aligned(16)));

(Although probably the real fix would be within the Arduino_TensorFlowLite
library to not do that check – does it make sense on these platforms?). 

Hope this helps :-3

ugmurthy, 2021-01-07:
I have the same problem when i run the classifier on nano33 BLE sense.
The com port is frozen and I cannot upload any new sketch. I restarted nano and also rebooted my laptop. The existing classifier on the nano prevent any new sketches from being loaded onto the nano therefore cannot test the suggestion made by @Rainycat

per1234, 2021-01-07:
Hi @ugmurthy. You can recover your board from the Mbed OS crash by:
1. Press and release the reset button on your board quickly twice. You should now see the "L" LED on the board pulsing, which means the bootloader is running. If you don't see the LED pulsing, you might not have gotten the timing of the button press right, so try again. The double reset causes the bootloader to run indefinitely (until the board is reset, powered off, or an upload is done).
1. Select the port of your board from the **Tools > Port** menu. The port number may be different when the bootloader is running so don't assume you already have the correct port selected.

Now you should be able to upload the sketch to your Arduino board.

ugmurthy, 2021-01-07:
> Just to update anyone else who is having this problem, after a lot of faff, we discovered that in MicroAllocator (called by MicroInterpreter instantiation), it checks the supplied arena is 16-byte [...]

Yes that helps - the classifier works! thank you

ab-cp, 2021-01-13:
> Hi @ugmurthy. You can recover your board from the Mbed OS crash by:
> 
> 1. Press and release the reset button on your board quickly twice. You should now see the "L" LED on the board pulsing, which means the bootloader is running. If you don't see the LED pulsing, you might not have gotten the timing of the button press right, so try again. The double reset causes the bootloader to run indefinitely (until the board is reset, powered off, or an upload is done).
> 2. Select the port of your board from the **Tools > Port** menu. The port number may be different when the bootloader is running so don't assume you already have the correct port selected.
> 
> Now you should be able to upload the sketch to your Arduino board.

Both this and 16 bit alignment works!

closed 2021-03-13 by KXCY-AI