arduino/ArduinoTensorFlowLiteTutorials #30: it can be apply to nano 33 iot?

MAKER-park, 2021-06-25:
hello!
i'm try to apply this code form nano 33 iot.
imu capture just change header name just 1->3

but imu classficate apear this error what the problem or just it is not support nano33iot imu sensor?

my eeror page like this

아두이노:1.8.15 (Windows 10), 보드:"Arduino NANO 33 IoT"

Library Arduino_TensorFlowLite has been declared precompiled:Precompiled library in "C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src\cortex-m0plus" not foundIn file included from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/Core/Include/cmsis_compiler.h:54:0,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math_types.h:76,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:199,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:33,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:164,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src\tensorflow\lite\micro\kernels\cmsis-nn\fully_connected.cpp:18:C:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:58:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE void __enable_irq(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:69:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE void __disable_irq(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:80:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_CONTROL(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEIn file included from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/Core/Include/cmsis_compiler.h:54:0,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math_types.h:76,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:199,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:33,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:164,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src\tensorflow\lite\micro\kernels\cmsis-nn\conv.cpp:19:C:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:58:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE void __enable_irq(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:94:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE void __set_CONTROL(uint32_t control)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:69:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE void __disable_irq(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:105:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_IPSR(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:119:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_APSR(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:134:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_xPSR(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:148:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_PSP(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:162:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE void __set_PSP(uint32_t topOfProcStack)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:173:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_MSP(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:188:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE void __set_MSP(uint32_t topOfMainStack)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:199:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE uint32_t __get_PRIMASK(void)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:213:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__( ( always_inline ) ) __STATIC_INLINE void __set_PRIMASK(uint32_t priMask)                                    ^~~~~~~~~~~~~~~                                    _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:373:32: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'? __attribute__((always_inline)) __STATIC_INLINE void __NOP(void)                                ^~~~~~~~~~~~~~~                                _ELIDABLE_INLINEC:\Users\ksh26\AppData\Local\Arduino15\packages\ar [... cut ...] 1_to_q7(   ^~~~~~~~~~~~~~~~~~~~In file included from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:211:0,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:33,                 from C:\Users\ksh26\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite

facchinm, 2021-06-28:
Hi @MAKER-park ,
cortex-m0 is not a first class target for TFLite library, so it's likely to be buggy / only working on some versions.
Which version of SAMD core and Tensorflow_lite libraries are you using?

MAKER-park, 2021-08-20:
> Hi @MAKER-park ,
> 
> cortex-m0 is not a first class target for TFLite library, so it's likely to be buggy / only working on some versions.
> 
> Which version of SAMD core and Tensorflow_lite libraries are you using?

Thank you!! For answer me i first know that model use difference chipset!

closed 2021-08-20 by aentinger