arduino/ArduinoTensorFlowLiteTutorials #31: TFLite on MKR1010 Wifi - Error from CMSIS

skye0402, 2021-08-23:
I tried to run the Sine-wave tutorial as well as the "magic-wand" example. Both error at the same point:
`In file included from src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/Core/Include/cmsis_compiler.h:54:0,
                 from src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math_types.h:76,
                 from src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:199,
                 from src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:33,
                 from src/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:164,
                 from src\tensorflow\lite\micro\kernels\cmsis-nn\conv.cpp:19:
ArduinoData\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/cmsis_gcc.h:58:36: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'?
 __attribute__( ( always_inline ) ) __STATIC_INLINE void __enable_irq(void)
                                    ^~~~~~~~~~~~~~~
                                    _ELIDABLE_INLINE
`

I'm using a MKR1010 WiFi module which runs Arm Cortex M0+. Any idea what it could be? It gets pretty far in the making from what I can tell.

per1234, 2021-08-23:
For some information, see the reply for a previous (closed by reporter even though unresolved) report of the same error here:
https://github.com/arduino/ArduinoTensorFlowLiteTutorials/issues/30#issuecomment-869451601

skye0402, 2021-08-23:
Thank you @per1234 , saw that yesterday. The BLE 33 it's a Cortex M0+ as well so wondering if there's a way around that error.

aentinger, 2021-08-30:
The Nano 33 BLE is a nRF52840  which is Cortex-M4F.

closed 2021-08-30 by aentinger