arduino/ArduinoTensorFlowLiteTutorials #32: Breakage caused by removal of "Arduino_TensorFlowLite" library from Library Manager

per1234, 2022-09-02:
The "**Arduino_TensorFlowLite**" library dependency of these tutorials was removed from Library Manager at the request of the TensorFlow Lite Micro maintainers (https://github.com/arduino/library-registry/pull/1748).

That removal impacts this project in several ways:

## Documentation

The installation instructions specify installation of the library from Library Manager:

https://github.com/arduino/ArduinoTensorFlowLiteTutorials/blob/master/GestureToEmoji/exercises/exercise1.md#arduino-libraries

This is no longer possible.

TensorFlow's library must be installed manually. The library's documentation provides instructions:

https://github.com/tensorflow/tflite-micro-arduino-examples#how-to-install

## Code

TensorFlow's library does not contain the file `tensorflow/lite/version.h`, so the code that contains an `#include` directive for this header no longer compiles:

```text
home/runner/work/ArduinoTensorFlowLiteTutorials/ArduinoTensorFlowLiteTutorials/GestureToEmoji/ArduinoSketches/IMU_Classifier/IMU_Classifier.ino:28:10: fatal error: tensorflow/lite/version.h: No such file or directory
   #include <tensorflow/lite/version.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
```

## CI

The repository contains a GitHub Actions workflow that provides a "smoke test" on relevant changes to the project by compiling the sketches:

https://github.com/arduino/ArduinoTensorFlowLiteTutorials/blob/master/.github/workflows/compile-sketches.yml

The workflow is [configured](https://github.com/arduino/compile-sketches#libraries) to install the dependency from Library Manager, which now fails:

```text
Running command: /home/runner/bin/arduino-cli lib install Arduino_TensorFlowLite 
   Error installing Arduino_TensorFlowLite: Library 'Arduino_TensorFlowLite' not found
```

whubaichuan, 2022-10-20:
@per1234 Hi, thanks for you finding the same problem. In my project which needs to use **Arduino_TensorFlowLite**
I just comment the code below:
```
\\#include <tensorflow/lite/version.h>
```
and it works. It seems no harm to the whole project.

Would you like to have a try here? Looking forward to your feedback.

per1234, 2022-10-20:
@BenjaminDannegard this information might be of interest to you ^^