arduino/ArduinoTensorFlowLiteTutorials #34: Compilation error: tensorflow/lite/micro/kernels/micro_ops.h: No such file or directory

vpranesh, 2023-02-08:
I am currently working with arduino nano 33 ble sense. Since the tensorflow lite library for arduino was removed, I manually installed using this link https://downloads.arduino.cc/libraries/github.com/bcmi-labs/Arduino_TensorFlowLite-1.15.0-ALPHA.zip?_gl=1*1yhwoez*_ga*NzU5Nzg3Mzg0LjE2NzUxNTU5MDk.*_ga_NEXN8H46L5*MTY3NTg4MDMwNi4xMC4xLjE2NzU4ODA0NjguMC4wLjA.

While trying to run my code I'm getting this error eventhough I have the files exactly in the path mentioned. The header files look like this

#include "TensorFlowLite.h"
#include "main_functions.h"
#include "tensorflow/lite/micro/kernels/micro_ops.h"
#include "tensorflow/lite/micro/micro_error_reporter.h"
#include "tensorflow/lite/micro/micro_interpreter.h"
#include "tensorflow/lite/micro/micro_mutable_op_resolver.h"
#include "tensorflow/lite/schema/schema_generated.h"
#include "tensorflow/lite/version.h"

Can someone please help me?

per1234, 2023-02-08:
Hi @vpranesh. Where did you find that link? It is to a very outdated version of the library. Your code was intended to be used with a recent version (i.e., 2.4.0-ALPHA). This is the cause of the error.

Please give it a try with 2.4.0-ALPHA. If you still have a problem, post on the Arduino Forum:

https://forum.arduino.cc/

We'll surely be able to help you out over there.

vpranesh, 2023-02-08:
Hey @per1234 I found the link in this Forum https://forum.arduino.cc/t/why-i-cant-find-the-arduino-tensorflowlite-library-in-library-management/1044070..

Can you please provide me with the link for the new version?
Thanks

vpranesh, 2023-02-08:
> Hey @per1234 I found the link in this Forum https://forum.arduino.cc/t/why-i-cant-find-the-arduino-tensorflowlite-library-in-library-management/1044070..
> 
> Can you please provide me with the link for the new version? Thanks

I found it..thanks

per1234, 2023-02-08:
>I found the link in this Forum https://forum.arduino.cc/t/why-i-cant-find-the-arduino-tensorflowlite-library-in-library-management/1044070..

Ouch, that was actually me :facepalm:, but in my defense it was in response to someone specifically requesting that version.

>I found it..thanks

You are welcome. If the problem is resolved you can close this issue.

We have a dedicated issue about updating the tutorials to work with TensorFlow's official Arduino Library now that Arduino's mirror library is no longer available: https://github.com/arduino/ArduinoTensorFlowLiteTutorials/issues/32

Mewankyrshan, 2023-05-11:
> 

Hey. Could you please attach the download link?